Xilinx/SDAccel_Examples #1: make all and make all-POWER problems

ErcanCanhasi, 2016-09-19:
Hi,

1) make all-POWER Launches compilation but produced executable hello_POWER.exe cannot be execute.
Error is: cannot execute binary file
2) make all produces hello_X86.exe
and error is: libOpenCl.so not found

I have tried many methods and getting emulation, hw or sw, is not a big deal when tlc scripts are used,   but executing opencl code on fpga is still very problematic. 
I have get stuck at the same problem even when I use run_system command in sdaccel.

Spenser309, 2016-10-06:
Hey ErcanCanhasi,

Can you list the full set of steps you are trying to execute?

With regards to "1" the binaries from the all-POWER builds are created for IBM Power PC platforms.  If you not running these on an IBM POWER platform than this is the expected behaviour.

With regards to "2" it appears you do not have libOpenCL.so in your ld-config or LD_LIBRARY_PATH.  To add it do the following

$ xbinst -x "<dsa>" -d . && mv xbinst/pkg/pcie/runtime .
$ export LD_LIBRARY_PATH=$PWD/runtime/lib/x86_64/:$LD_LIBRARY_PATH
$ export XILINX_OPENCL=$PWD

# For HW and SW Emualtion
$ export LD_LIBRARY_PATH=$XILINX_SDACCEL/runtime/lib/x86_64:$XILINX_SDACCEL/lib/lnx64.o
$ export XILINX_OPENCL=$XILINX_SDACCEL
$ export XCL_EMULATION_MODE=true

Spenser309, 2016-12-19:
closed due to lack of response.

closed 2016-12-19 by Spenser309