Xilinx/SDAccel_Examples #14: Suggestion to improve EoU of the current makefiles in order to provide the customers more flexibility and guidences

sergeisxilinx, 2016-11-02:
The goal of the current makefile changes is to improve EoF of the GitHub designs. Using the median_filter example I modified several makefiles to demonstrate these changes.

By modifying the example I was trying to achieve several goals:
- To limit the amount of changes the current makefile structure
- The script can be easily adopted for any design we have in GitHub
- **Work Directory**:
Currently the user should run the design from the directory where the design sources are located. For instance vision/median_filter. This is not convenient, because if you wish to remove all generated files and start from scratch, you have a risk to delete the original files. Therefore, now you can now run the design from the work directory, located under vision/median_filter.
o	Note: if you look into the work directory you will see the xclbin directory (I suppose that it is probably used also for testing purposes). If you delete it, the make flow will automatically recreate it from scratch.

- Using the provided makefile you can:
    o) Only compile
    o) Only run (after the application is compiled)
    o) Both compile and run 
the application. 

- In addition, using the existing makefile variable TARGETS, you can specify if you wish to run the application for 
    o) CPU emulation: TARGETS=sw_emu
    o) Hardware Emulation: TARGETS=hw_emu
    o) FPGA Deployment: TARGETS=hw

- I also added help information about the script:
   Run the following command from the work directory
   make -f ../Makefile info

- If you only run the makefile for compilation, then the makefile will suggest you how to executed the application.
Here are some examples:
    o) You compile the design for CPU Emulation:
         make -f ../Makefile TARGETS=sw_emu compile
         The compilation step will create all necessary files to run the sw_emu and provide the following info

==================================================================================================
Tip: to run the application in a 'sw_emu' mode you may use one of the following methods

   1) execute the makefile using 'run' as a target
          make -f ../Makefile <variables> run

      Note: if you wish to generate a TRACE view, then in addition, you need set TRACE=yes
          make -f ../Makefile TRACE=yes <variables> run

   2) run the following 2 commands
          export XCL_EMULATION_MODE=true
          ./median_X86.exe ../data/<bmp_file> xclbin/krnl_median.sw_emu.xilinx_adm-pcie-ku3_2ddr_3_1.xclbin

      Note: if you wish to generate a TRACE view, then in addition, you need set the following
            variables before running the application
          export SDACCEL_TIMELINE_REPORT=true
          export SDACCEL_DEVICE_PROFILE=true
==================================================================================================

Please see the Median_Filter.docx file for more info about the changes.
[Median_Filter.docx](https://github.com/Xilinx/SDAccel_Examples/files/567806/Median_Filter.docx)
[NEW_Median_Filter.tar.gz](https://github.com/Xilinx/SDAccel_Examples/files/567807/NEW_Median_Filter.tar.gz)

Spenser309, 2016-12-19:
For emulation, it's just as easy to set these in the environment as defining them on the command line.  Documentation updates to identify this would be useful.  With regards to runs on hardware, the nimbix-run.py script can be updated to look for the environment variables and set the run accordingly.  This would make the interface more uniform.  I'll leave this open and work on that implementation.

With regards to the "run" target, this has been implemented as "check" as this is the more common name of this target in the software community.

Spenser309, 2016-12-19:
reopened for nimbix-run.py improvements.

Spenser309, 2017-03-06:
Closed.  sdaccel.ini file can be used for enabling debugging and profiling even on Nimbix.

closed 2017-03-06 by Spenser309