Xilinx/SDAccel_Examples #16: not support multiple devices in the platform

alare, 2017-04-19:
In the libs/xcl/xcl.c file, the function call "xcl_world_single_vendor" scan the platform ID to match the vendor name "Xilinx", but do not scan the device ID with the same platform name "Xilinx".

The codes only call the function "clGetDeviceIDs" once to get the first device, then use the device name to match the xclbin file. 
1. So it always use the first device even through there are multiple FPGA cards with different Xilinx chips.
2. The codes are too ugly because it relies on the xclbin file name to match the device name.  There are some embedded text information inside the xclbin file, why not use these information instead?

I know the xcl.c is neither part of SDAccel standard nor OpenCL specs. But since it is inside the official git hub, many people will use these wrapper functions instead of OpenCL functions when they are using SDAccel.

Spenser309, 2017-04-26:
> In the libs/xcl/xcl.c file, the function call "xcl_world_single_vendor" scan the platform ID to match the vendor name "Xilinx", but do not scan the device ID with the same platform name "Xilinx".

This is intended to be done by https://github.com/Xilinx/SDAccel_Examples/blob/master/libs/xcl/xcl.c#L175 .  Can you elaborate on the bug you are seeing?

> The codes only call the function "clGetDeviceIDs" once to get the first device, then use the device name to match the xclbin file.

> So it always use the first device even through there are multiple FPGA cards with different Xilinx chips.

Actually, in the current release of SDAccel only homogeneous cards are supported.

> The codes are too ugly because it relies on the xclbin file name to match the device name. There are some embedded text information inside the xclbin file, why not use these information instead?

To do this we would need to scan for all xclbins and then ask "are you my kernel"?  This is may be possible; however, I have chosen to use this simpler scheme.

> I know the xcl.c is neither part of SDAccel standard nor OpenCL specs. But since it is inside the official git hub, many people will use these wrapper functions instead of OpenCL functions when they are using SDAccel.

The intention of xcl.c is to make the examples clear and understandable.  It does not cover all use cases, and is not intended for performance.

Spenser309, 2017-06-08:
Due to lack of response.  I'm going to close this out.

closed 2017-06-08 by Spenser309

mathmax12, 2017-11-17:
@Spenser309 
Hi, I am looking for how to leverage multiple devices in a signal device in SDAccel.

Do you have a such kind of example to read?

Thanks 
Sean

DutchAlthoff, 2017-11-17:
Hi @mathmax12 

Please post your question on the [Xilinx SDAccel forums](https://forums.xilinx.com/t5/SDAccel/bd-p/SDx). 

This ticket is closed and should only be used as a reference.