Xilinx/SDAccel_Examples #18: clEnqueueWriteBuffer() cause DMA deadlock when transfering large dataset

doonny, 2017-08-02:
In the example "dataflow_pipes_ocl", It is said in the host code:

// Using clEnqueueMigrateMemObjects() instead of clEnqueueWriteBuffer() to avoid
// deadlock in real hardware which can be noticed only for large dataset.
// Rootcause: design leads to a deadlock when host->DDR and 
// output_stage->DDR causes a contention and deadlock. In small dataset, the 
// data gets transferred from host-> DDR in 1 burst and hence no deadlock.
**// Solution: Start output_stage when host->DDR data transfer is completed.
// clEnqueueMigrateMemObject() event is used for all three kernels to avoid deadlock.**

Why clEnqueueWriteBuffer() does not work for large dataset ?  One could also use blocking Writebuffer and wait for the transfer of the large dataset is complete, Since it is said in the comments as long as  host->DDR data transfer is completed, there will be no contention may exits.

heeran-xilinx, 2017-08-06:
Hi @doonny,
clEnqueueWriteBuffer() API does not immediately copy data to device. It just copy write to a buffer object from host memory. Incase of blocking call also, it just guaranteed that host memory to buffer object copy is finished. But data is still not copied to device, it just copied to buffer object.
Actually data-copy will happen when some kernel is started which has respective buffer object in kernel argument list. This architecture can create a deadlock situation for large dataset. To avoid this deadlock, it is recommended to user clEnqueueMigrateMemObject() API which does immediate data transfer from host to device.

doonny, 2017-08-17:
@heeran-xilinx Thx for the reply.
If clEnqueueWriteBuffer() can cause a deadlock, I think the same may happen for clEnqueueReadBuffer() ? So why do you use clEnqueueReadBuffer() in the example code instead of clEnqueueMigrateMemObject() again ?

heeranand, 2017-09-13:
Hi @doonny , first of all, sorry for late reply.

We have seen deadlock for example "getting_started/dataflow/dataflow_pipes_ocl" for large data-set. Using migrateMemObj API, this deadlock is avoided. 

For this example, all three kernels triggered concurrently (input, adder, and output) from host using OUT_OF_ORDER queue. input kernel has dependency on input buffer. So until input buffer is transferred completely, it will not start input kernel. Whereas adder and write kernels have no such dependency, so these kernels will start immediately.  So write kernel (3rd kernel)  will initiate write request to DDR with the hope that data will be available in pipe P2 soon to transfer. But P2 will never get data until read kernel starts its operation. So it creates a deadlock situation. 

I hope it will clarify your doubt.

-Heera

doonny, 2017-09-15:
Hi @heeranand really thanks for your reply.

Now I see how the deadlock is caused. However, personally, I do not think this kind of situation should happen.  What I mean is that the DDR controller should be able to process another read request from the input kernel, even when the write request from the output kernel could not be processed, i.e., an unprocessed write request operation should not cause a "deadlock", it can cause the pipeline to stall but never a deadlock. Independent channels in DDR controller should be able to handle such operations.

Otherwise, this will limit the application and flexibility of the concept of "out of order queues" that is initially introduced by OpenCL.

heeranand, 2017-09-15:
Hi @doonny ,
Read and write request can be served concurrently by DDR controller. But two concurrent write request cannot be served by DDR controller.   
This deadlock case is specific to two writes to DDR.  First write is from host memory to DDR (buffer copy) and 2nd write is from output kernel to DDR. 
But I completely agree that this situation should be handled by DSA/tool properly to avoid deadlock. This will be addressed in future release.
 
-Heera

closed 2017-09-16 by doonny