Xilinx/SDAccel_Examples #19: Verification of gemm fails

definelicht, 2017-08-02:
Verification of `acceleration/gemm` fails on xilinx:xil-accel-rd-ku115:4ddr-xpr:4.0 with SDx 2017.1.

Example execution below:
```
% LD_LIBRARY_PATH=/opt/Xilinx/SDx/2017.1/runtime/lib/x86_64 ./gemm 4096 4096 64
Creating context...
INFO: Importing xclbin/gemm0.hw.xilinx_xil-accel-rd-ku115_4ddr-xpr.xclbin
INFO: Loaded file
INFO: Created Binary
INFO: Built Program
INFO: input matrix size: M= 4096, N= 4096, K= 64
INFO: hw matrix size: row= 128, col= 64, depth= 64
Creating Buffers...
Copying Buffers to device....
Copying results to host....
INFO: Execution done
ERROR in - 0 - actual=31, expected=0
INFO: kernel time 0.004075 seconds numOfOps -2147483648.000000 Efficiency: -526.929595 GOPs
INFO: Test Failed
```

Steps to reproduce:
1. `make all TARGETS=hw`
2. `.LD_LIBRARY_PATH=/opt/Xilinx/SDx/2017.1/runtime/lib/x86_64 /gemm 4096 4096 54`

Installed DSA: xilinx:xil-accel-rd-ku115:4ddr-xpr:4.0

Other kernels execute correctly.

What are the acceptable parameters for rows, cols and depth, and what is the effect of depth? Perhaps put this in the readme?

xkang79, 2017-08-02:
the parameters we used are 64 64 64

definelicht, 2017-08-02:
Does that translate to a 64x64 matrix? Does it not support larger matrices?
Or does this relate to the hardware buffer?

It would be very useful if you could add some documentation on the input parameters :-)

xkang79, 2017-08-02:
limitations on parameters:
    - M : multiple of 32 &  <= 32736
    - N : multiple of 64  &  <= 65472
    - K : multiple of 64   &  <= 16320

definelicht, 2017-08-03:
The three upper bounds you listed are not multiples of 32/64, so I suppose it should be strictly lower than?

The kernel does not verify for me for any other parameters than 64 64 64.
128 128 64 yields wrong results.
Is this expected?

troore, 2017-08-26:
Could the developers also provide the source files for kernels? As a newbie for SDAccel tools, I encountered several device does not match issues. Maybe I want to try to compile the kernel parts myself. Thanks.

heeran-xilinx, 2017-08-30:
Hi @definelicht  and @troore  we will get back to your question.
I can answer @troore quesiton: how to put two kernel functions into one program using C/C++ rather than OpenCL? 
Please refer below example ( it has two C/C++ kernels inside single binary container):
https://github.com/Xilinx/SDAccel_Examples/tree/master/getting_started/kernel_opt/loop_perfect_c

xkang79, 2017-08-30:
hi @troore @definelicht , we can't provide source code for it. It is provided as a block and is encrypted. 

I tried the parameters other than 64 64 64, e.g. 128 128 64 it works on my system. I guess it is the issue of the sdaccel set up that causes the failure.

DutchAlthoff, 2017-08-30:
@definelicht I too have not been able to reproduce this issue. Please email me with further questions.

@troore This is an RTL kernel optimized for the xil-accel-rd-ku115:4ddr-xpr device. 
Please search/post in the [Xilinx SDAccel forums](https://forums.xilinx.com/t5/SDAccel/bd-p/SDx) for help with your SDAccel environment setup.
For general SDAccel questions please refer to the [SDAccel docs](https://www.xilinx.com/html_docs/xilinx2017_1/sdaccel_doc/index.html). 
Regarding your C/C++ question see the SDAccel docs [page here](https://www.xilinx.com/html_docs/xilinx2017_1/sdaccel_doc/topics/design-flows/concept-express-kernel-c-cplusplus.html)

closed 2017-08-30 by DutchAlthoff