Xilinx/SDAccel_Examples #20: Failed to Run Examples

zainryan, 2017-08-09:
Hi all,

I failed to run any examples here and I guess it's due to the opencl runtime environment. 
My hardware platform is xilinx:adm-pcie-7v3:1ddr:3.0 and software platform is SDx-2017.1.
Taking /getting_started/host/helloworld_ocl for instance, If I compile host program with linkage -lOpenCL, executing it will immediately complain **Error: Failed to find Xilinx platform**.
If I compile host program with linkage -lxlinxopencl, the situation seems a little bit better. The output of host program is shown below.

**platform Name: Xilinx
Vendor Name : Xilinx
Found Platform
Found Device=xilinx:adm-pcie-7v3:1ddr:3.0
XCLBIN File Name: vector_addition
INFO: Importing xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin
Loading: 'xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin'
ERROR: Failed to load xclbin
ERROR: program is nullptr
[1]    3626 segmentation fault (core dumped)  ./helloworld**

Does anyone have any advice? Your help is really appreciated.

zainryan, 2017-08-09:
The complete command to compile my host program is: **xcpp -Wall -O0 -g -I./src/ -I/opt/Xilinx/SDx/2017.1/runtime/include/1_2 -I../../../libs/xcl2  -std=c++0x -o helloworld ./src/host.cpp  ../../../libs/xcl2/xcl2.cpp -L/opt/Xilinx/SDx/2017.1/runtime/lib/x86_64 -L/opt/Xilinx/SDx/2017.1/lib/lnx64.o -lOpenCL -pthread**

Changing -lOpenCL to -lxilinxopencl make the case a little bit better.

zainryan, 2017-08-09:
In addition, if I compile getting_started/host/device_query_ocl with -lxilinxopencl, device query passes the test. The output is shown below.

**platform profile    : EMBEDDED_PROFILE
platform version    : OpenCL 1.0
platform name       : Xilinx
platform vendor     : Xilinx
platform extensions : cl_khr_icd
Device 0: xilinx:adm-pcie-7v3:1ddr:3.0
  device type                            : CL_DEVICE_TYPE_ACCELERATOR
  device vendor id                       : 0
  device max compute units               : 1
  device max work item dimensions        : 3
  device max work group size             : 4096
  device max work item sizes             : [ 4096 4096 4096 ]
  device preferred vector width char     : 1
  device preferred vector width short    : 1
  device preferred vector width int      : 1
  device preferred vector width long     : 1
  device preferred vector width float    : 1
  device preferred vector width double   : 0
  device max clock frequency             : 200
  device address bits                    : 32
  device max read image args             : 128
  device max write image args            : 8
  device max mem alloc size              : 536870912
  device image2d max width               : 8192
  device image2d max height              : 8192
  device image3d max width               : 2048
  device image3d max height              : 2048
  device image3d max depth               : 2048
  device image support                   : true
  device max parameter size              : 2048
  device max samplers                    : 0
  device mem base addr align             : 32768
  device min data type align size        : 128
  device single fp config                : [ CL_FP_INF_NAN ]
  device global mem cache type           : CL_NONE
  device global mem cacheline size       : 64
  device global mem cache size           : 0
  device global mem size                 : 8589934592
  device max constant buffer size        : 4194304
  device max constant args               : 8
  device local mem type                  : CL_GLOBAL
  device local mem size                  : 16384
  device error correction support        : true
  device profiling timer resolution      : 1
  device endian little                   : true
  device available                       : false
  device compiler available              : true
  device execution capabilities          : [ CL_EXEC_KERNEL ]
  device queue properties                : [ CL_QUEUE_OUT_OF_ORDER_EXEC_MODE_ENABLE CL_QUEUE_PROFILING_ENABLE ]
  device name                            : xilinx:adm-pcie-7v3:1ddr:3.0
  device vendor                          : Xilinx
  device version                         : 1.0
  device profile                         : EMBEDDED_PROFILE
  device version                         : OpenCL 1.0
  device extensions                      : 
  device platform                        : Xilinx
  device double fp config                : [ ]
  device preferred vector width half     : 1
  device host unified memory             : true
  device native vector width char        : 1
  device native vector width short       : 1
  device native vector width int         : 1
  device native vector width long        : 1
  device native vector width float       : 1
  device native vector width double      : 1
  device native vector width half        : 1
  device opencl c version                : OpenCL C 1.0
  device linker available                : true
  device built in kernels                : 
  device image max buffer size           : 65536
  device image max array size            : 2048
  device parent device                   : N/A 
  device partition max sub devices       : 0
  device partition properties            : NONE
  device partition affinity domain       : [ ]
  device partition type                  : N/A 
  device reference count                 : 1
  device preferred interop user sync     : true
  device printf buffer size              : 0
TEST PASSED**

heeranand, 2017-08-14:
**First issue:** 
while using -LOpenCL, it searches vendor specific icd file at following location:
/etc/OpenCL/vendors
So xilinx.icd file should present there. 
For more details, please refer this:
https://www.xilinx.com/html_docs/xilinx2017_1/sdaccel_doc/topics/platform-mem-model/con-opencl-icd-loader.html

**Second issue:**
Loading: 'xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin'
ERROR: Failed to load xclbin
can you please cross check that respective xclbin file exist in xclbin repository?

-Heera

zainryan, 2017-08-14:
Thank you, Herra! Now I know the difference of -LOpenCL and -Lxilinxopencl.
However, for the second issue, I've double checked that this xclbin file does exist in the right path. 

```
➜  helloworld_ocl git:(master) ✗ file xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin
xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin: data
➜  helloworld_ocl git:(master) ✗ ./helloworld 
platform Name: Xilinx
Vendor Name : Xilinx
Found Platform
Found Device=xilinx:adm-pcie-7v3:1ddr:3.0
XCLBIN File Name: vector_addition
INFO: Importing xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin
Loading: 'xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin'
ERROR: Failed to load xclbin
ERROR: No program executable for device
ERROR: buffer (2) is not resident in device (0)
Result = 
Error: Result mismatch:
i = 0 CPU result = 42 Device result = 0
TEST FAILED
terminate called after throwing an instance of 'xrt::error'
  what():  event 1 never submitted
[1]    8362 abort (core dumped)  ./helloworld
```
And I've used gdb to locate the error point. It's on line 67, cl::Program program(context, devices, bins);
```(gdb) n
INFO: Importing xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin
Loading: 'xclbin/vector_addition.hw.xilinx_adm-pcie-7v3_1ddr.xclbin'
66	    devices.resize(1);
(gdb) n
67	    cl::Program program(context, devices, bins);
(gdb) n
ERROR: Failed to load xclbin
74	            size_in_bytes, source_a.data());
```

zainryan, 2017-08-14:
Is it possible that old driver is the root cause? The driver of this machine installed by admin is the corresponding one in SDAccel 2016.1.

heeranand, 2017-08-14:
Yes. It could be older driver issue. Actually example host code is migrated to khronos OpenCL C++ which was not there in 2016.1 release. Please install the latest driver and try again.

closed 2017-08-30 by heeran-xilinx