Xilinx/SDAccel_Examples #21: Communication between SLRs

troore, 2017-08-31:
In this example, if I want to make the two kernels executing on two SLRs communicate with each other, which mechanisms should I use to make it happen?

Thanks.

akasat, 2017-09-13:
How do you expect the two kernsl to communicate. Is it a direct pipe connection, or is it through DDR. If its through DDR, then atleast one of the kernels will need to be configured to have multiple AXI interfaces and map different kernel arguments to different DDRs. The AXI Interconnect between the kernel and the DDR will span across SLRs.

troore, 2017-10-26:
@akasat Thanks for your explanation. After some surveying, I learned that Xilinx synthesis tool is able to do cross SLR placement and routing. And your suggested methods are also helpful.

closed 2017-10-26 by troore