Xilinx/SDAccel_Examples #22: "no HwEm HAL handle" error when executing hardware emulation

troore, 2017-09-03:
Hi,

When I compile and run gemm in hw_emu mode, I encounter a crash error in clCreateProgramWithBinary:

`
ERROR: device::load_binary binaary target=HwEm, no HwEm HAL handle
`
`
Error: Failed to create compute program from binary -44!
`

What causes this error? Thanks.

troore

akasat, 2017-09-04:
Hi,

This could happen In following cases
1. hw_em driver is not present at <sdx>/data/emulation/hw_em/generic_pcie/driver/ location
2. hw_em driver is not loaded properly by runtime

Can you confirm that your installation is fine? Are you able to run SW emulation.

troore, 2017-09-04:
@akasat Thanks for your help! I've found the reason. That is very simple, I forgot to turn on the XCL_EMULATION_MODE. But I still found the reason according to your advice as the emconfigutil cannot change clGetDeviceInfo for device name when running.

closed 2017-09-04 by troore