Xilinx/SDAccel_Examples #23: Cannot upgrade to invalid target ''

inthesunxkz, 2017-09-16:
when I make my rtl logic, I get the error:
ERROR: [XOCC 19-98] Generation of the IP CORE failed.
Cannot upgrade to invalid target ''
How should I to solve it? Thanks.

akasat, 2017-09-17:
Hi,
Which version of SDAccel are you running the test with. And what is the target platform? Can you share the full makefile command you are using?

Regards,
Amit

inthesunxkz, 2017-09-19:
@akasat Hi!
The version of SDaccel is 2017.1 and the platform is adm-pcie-7v3.
Thanks

heeran-xilinx, 2018-02-06:
Hi @inthesunxkz ,
I would suggest if you can send your query to Xilinx SDAccel forum?
https://forums.xilinx.com/t5/SDAccel/bd-p/SDx

There you may get quick support as xilinx Technical support team is actively following the forum queries. 

-Heera

closed 2018-02-06 by heeran-xilinx