Xilinx/SDAccel_Examples #25: Child process did not launch

Rpona, 2017-11-08:
Hi,

I met this problem and I found that kcu1500 isn't in the support device list.
Is it the reason for this error message?   
If yes, could you help me to fixed it? or give me some suggestions to fixed it?
Thank you.  

--  SDAccel_Examples/getting_started/clk_freq/large_loop_ocl --
INFO: Importing xclbin/cnn_GOOD.sw_emu.xilinx_kcu1500_4ddr-xpr_4_0.xclbin
INFO: Loaded file
FATAL ERROR : child process did not launch
WARNING: Profiling may contain incomplete information. Please call clReleaseProgram() from your host code.
WARNING: Profiling may contain incomplete information. Please call clReleaseProgram() from your host code.
Segmentation fault (core dumped)

-- SDAccel_Examples/getting_started/host/helloworld_ocl --
platform Name: Intel(R) FPGA SDK for OpenCL(TM)
Vendor Name : Xilinx
platform Name: Xilinx
Vendor Name : Xilinx
Found Platform
Found Device=xilinx_kcu1500_4ddr-xpr_4_0
XCLBIN File Name: vector_addition
INFO: Importing xclbin/vector_addition.sw_emu.xilinx_kcu1500_4ddr-xpr_4_0.xclbin
Loading: 'xclbin/vector_addition.sw_emu.xilinx_kcu1500_4ddr-xpr_4_0.xclbin'
FATAL ERROR : child process did not launch
WARNING: Profiling may contain incomplete information. Please call clReleaseProgram() from your host code.
WARNING: Profiling may contain incomplete information. Please call clReleaseProgram() from your host code.
*** Error in `./helloworld': Segmentation fault (core dumped)

DutchAlthoff, 2017-11-08:
Hi @Rpona 

Are you using the check target? That is:
`make check TARGETS=sw_emu DEVICES=xilinx:kcu1500:4ddr-xpr`

For the large loop example comment out [this line](https://github.com/Xilinx/SDAccel_Examples/blob/81e002969a935250176a29f7820cf62ffed901e0/getting_started/clk_freq/large_loop_ocl/Makefile#L37)

Rpona, 2017-11-08:
Hi @DutchAlthoff 

If I used this commend:
_make check TARGETS=sw_emu DEVICES=xilinx:kcu1500:4ddr-xpr1_
or
_make TARGETS=sw_emu check_
**result:**
make: Nothing to be done for `check'.

-- My used commands are as below --
make TARGETS=sw_emu DEVICES=xilinx:kcu1500:4ddr-xpr:4.0 all
export LD_LIBRARY_PATH=$XILINX_SDX/runtime/lib/x86_64/:$LD_LIBRARY_PATH
export XCL_EMULATION_MODE=sw_emu
emconfigutil --xdevice 'xilinx:kcu1500:4ddr-xpr:4.0' --nd 1
**result:**
INFO: Importing xclbin/cnn_GOOD.sw_emu.xilinx_kcu1500_4ddr-xpr_4_0.xclbin
INFO: Loaded file
FATAL ERROR : child process did not launch
WARNING: Profiling may contain incomplete information. Please call clReleaseProgram() from your host code.
WARNING: Profiling may contain incomplete information. Please call clReleaseProgram() from your host code.
Segmentation fault (core dumped)

DutchAlthoff, 2017-11-08:
There is something is not right in your environement. 

For large loop, see my previous comment:

> For the large loop example comment out [this line](https://github.com/Xilinx/SDAccel_Examples/blob/81e002969a935250176a29f7820cf62ffed901e0/getting_started/clk_freq/large_loop_ocl/Makefile#L37)

Next:
Have you installed the [xilinx.icd](https://www.xilinx.com/html_docs/xilinx2017_1/sdaccel_doc/topics/platform-mem-model/con-opencl-icd-loader.html?hl=icd)?

Rpona, 2017-11-08:
Hi @DutchAlthoff 

Thanks for your reply.
Now, this command can work in the large_loop_ocl project.
However, it still met the same problem in the terminal.
--
I was already installed the xilinx.icd.
--
BTW, both of them can work well with this command "make TARGETS=**hw_emu** check" in the terminal.
Software emulation and Hardware emulation can run well with SDAccel GUI.

DutchAlthoff, 2017-11-18:
I am closing this issue because it is an environment configuration issue and not an issue with the examples.

[Please follow this guide](https://www.xilinx.com/html_docs/xilinx2017_2/sdaccel_doc/topics/xilinx-opencl-compiler/task-running-cpu-and-hardware-emulation-in-xocc-flow.html?hl=software%2Cemulation) for SDAccel emulation environment setup. 

If you have further questions about SDAccel please post them on the [Xilinx SDAccel forum](https://forums.xilinx.com/t5/SDAccel/bd-p/SDx).

closed 2017-11-18 by DutchAlthoff