Xilinx/SDAccel_Examples #26: Can I apply it on SDSoc based on ZCU102

kouyss, 2018-01-18:
Hi! I got two problems.
1. Can I apply it on SDSoc based on ZCU102
2.Following the UG1021, I choose zcu102（I don't have a KCU1500 and SDAccel license）, and the software platform are as follows:System configuration:A53 OpenCL Linux;Runtime:OpenCL（make it more similar to the kcu1500）,  and there was a problem in adding SDx Examples, I press the button "Download" for the SDAccel Examples(or the SDSoc Examples), An error windows popped up,“Download Error”:Couldn't download https://github.com/Xilinx/SDAccel_Examples.git.
Do you know the solution?
Regards

dutchalthoff, 2018-01-18:
@kouyss 
Which version of SDx are you running?

heeran-xilinx, 2018-01-18:
Hi @kouyss ,

Most of SDAccel examples are not directly compatible to SDSoC platform.  Xilinx is maintaining separate Git repository for SDSoC Specific example, please refer below link:
https://github.com/Xilinx/SDSoC_Examples
Note: OpenCL specific designs are placed under "ocl" folder. These examples are also downloadable from SDx Gui.

As Dutch also requested, what tool version are you using? There were some issues with previous version of tools (2017.1/.2) which are specific to Git Version.  can you try the latest tool 2017.4 sdx?

-Heera

kouyss, 2018-01-18:
@dutchalthoff @heeran-xilinx
Actually I used the latest tool 2017.4 sdx, and I encountered the same problem when I press "download" button of SDSoc_Examples.

closed 2018-02-06 by heeran-xilinx