Xilinx/SDAccel_Examples #31: 2017.1.RTE not availble after following the tutorial (Create, configure and test an AWS F1 instance)

mkhaled87, 2018-05-23:
In section [4: Running the SDAccel 'Hello World' example on AWS F1](https://github.com/Xilinx/SDAccel_Examples/wiki/Create,-configure-and-test-an-AWS-F1-instance#4-running-the-sdaccel-hello-world-example-on-aws-f1), in the part where you need to [execute the host application to run on the FPGA](https://github.com/Xilinx/SDAccel_Examples/wiki/Create,-configure-and-test-an-AWS-F1-instance#execute-the-host-application):

The link '/opt/Xilinx/SDx/2017.1.rte/setup.sh' is not valid since (2017.1.rte) is not present after following all previous steps with the latest (FPGA Developer AMI) from Amazon.

I tried to use the availble RTE which is (2017.4.rte.dyn) and the result is the app cant find a device and here is what it shows:

[0]user:0x1042:0x7:[???:??:0]
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
[0]user:0x1042:0x7:[???:??:65535]
device[0].user_instance : 65535
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
[0]user:0x1042:0x7:[???:??:0]
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
[0]user:0x1042:0x7:[???:??:65535]
device[0].user_instance : 65535
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
[0]user:0x1042:0x7:[???:??:0]
xclProbe found 1 FPGA slots with xocl driver running
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD0
WARNING: AwsXcl isGood: invalid user handle.
WARNING: xclOpen Handle check failed
[0]user:0x1042:0x7:[???:??:65535]
device[0].user_instance : 65535
WARNING: AwsXcl - Cannot open userPF: /dev/dri/renderD65535
WARNING: AwsXcl isGood: invalid user handle.
ERROR: xclOpen Handle check failed
ERROR: Device setup failed
Error: Failed to find Xilinx platform

lyager, 2018-06-04:
Same issue, running on latest AMI in AWS. Subscribing and hoping for solution.

kristopk, 2018-06-04:
Hi,
The latest aws-fpga has an ERRATA item:
Linux kernel 3.10.0-862.2.3.el7.x86_64. By default, the AWS Developer AMI GUI setup script updates the kernel version. We have provided a patch to prevent kernel updates during GUI setup. Instead of running the setup_gui.sh as documented/included within the developer AMI, please use the patched script as shown below: $curl https://s3.amazonaws.com/aws-fpga-developer-ami/1.4.0/Scripts/setup_gui.sh -o /home/centos/src/scripts/setup_gui.sh $/home/centos/src/scripts/setup_gui.sh

Another possible issue could be that you are using the incorrect AMI/github version:
This release supports Xilinx SDx 2017.4 and 2017.1.  The compatibility table describes the mapping of developer kit version to [FPGA developer AMI](https://aws.amazon.com/marketplace/pp/B06VVYBLZZ) version:  

| Developer Kit Version   | Tool Version Supported     |  Compatible FPGA developer AMI Version     |
|-----------|-----------|------|
| 1.3.0-1.3.6 | 2017.1 | v1.3.5 |
| 1.3.7-1.3.X | 2017.1 | v1.3.5-v1.3.X (Xilinx SDx 2017.1) |
| 1.3.7-1.3.X | 2017.4 | v1.4.0-v1.4.X (Xilinx SDx 2017.4) |

The [**Amazon FPGA Development User Forum**](https://forums.aws.amazon.com/forum.jspa?forumID=243&start=0) is the first place to go to post questions, learn from other users and read announcements from the EC2 FPGA team.
Thanks
Kris

lyager, 2018-06-06:
@kristopk thanks for the reply - I wasn't aware of the Amazon FPGA Development User Forum, I'll have a look. Do you have any more info about the ERRATA? I haven't run the setup_gui.sh script on my AWS instance, only the sdaccel_setup.sh script. My kernel version is `3.10.0-693.21.1.el7.x86_64` and the AMI version is 1.4.0.

Thanks in advance
/Jesper

jimw567, 2018-06-11:
Please check this post: https://forums.aws.amazon.com/thread.jspa?threadID=283820&tstart=0

ted-xie, 2018-06-28:
@lyager On my end, the setup.sh script is usually located under /opt/Xilinx/SDx/2017.1.rte**.4ddr**. Have you tried that directory yet? It always works for me.

lyager, 2018-06-29:
Thanks for the update @ted-xie it was an error in AMI-1.4.0 which required me to program the FPGA with another non-related image. After that, the host program discovered that the FPGA had wrong AGFI, and uploaded the correct one - and the it all worked again. 

Another thread here: https://github.com/aws/aws-fpga/issues/409

heeran-xilinx, 2018-06-29:
Thanks @ted-xie, @jimw567 and @kristopk for suggestion to resolve this issue. Closing this issue as it is resolved now. 
Please post AWS related queries to AWS forum in future.

closed 2018-06-29 by heeran-xilinx