Xilinx/SDAccel_Examples #32: Error hw_emu.

ramonn76, 2018-05-23:
Error running the following command: make check TARGETS=hw_emu DEVICES=$AWS_PLATFORM all.

How can I fix it? Any ideas?

Thanks.

****** vpl v2017.4 (64-bit)
  **** SW Build 2193837 on Tue Apr 10 18:06:59 MDT 2018
    ** Copyright 1986-2017 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file 'kernel_info.dat'.
INFO: [VPL 60-895]    Target platform: /home/centos/src/project_data/aws-fpga/SDAccel/aws_platform/xilinx_aws-vu9p-f1_dynamic_5_0/xilinx_aws-vu9p-f1_dynamic_5_0.xpfm
INFO: [VPL 60-423]   Target device: xilinx_aws-vu9p-f1_dynamic_5_0
INFO: [VPL 60-251]   Hardware accelerator integration...
ERROR: [VPL 60-777] Sorry, but it appears that a Xilinx program has terminated unexpectedly. Please contact Xilinx technical support for further assistance and give them the contents of /home/centos/src/project_data/aws-fpga/SDAccel/examples/xilinx_2017.4/getting_started/host/helloworld_ocl/_xocc_link_vector_addition.hw_emu.xilinx_aws-vu9p-f1_dynamic_5_0_vector_addition.hw_emu.xilinx_aws-vu9p-f1_dynamic_5_0.dir/_vpl/ipi/CrashLog
ERROR: [VPL 60-399] vivado failed, please see log file for detail: '/home/centos/src/project_data/aws-fpga/SDAccel/examples/xilinx_2017.4/getting_started/host/helloworld_ocl/_xocc_link_vector_addition.hw_emu.xilinx_aws-vu9p-f1_dynamic_5_0_vector_addition.hw_emu.xilinx_aws-vu9p-f1_dynamic_5_0.dir/_vpl/ipi/vivado.log'
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [XOCC 60-398] vpl failed
ERROR: [XOCC 60-626] Kernel link failed to complete
ERROR: [XOCC 60-703] Failed to finish linking

vishnuchebrolu, 2018-06-14:
Hi,
I tested this locally and I don't see any issue. 
[image]

kristopk, 2018-06-14:
Hi
Check the available disk space.  I have seen the same generic error when I forgot to size up the developer AMI volumes.
Kris

jimw567, 2018-06-14:
The other possibility is that the instance ran out of memory. What is the instance type you used for the FPGA AMI?

heeran-xilinx, 2018-06-29:
Thanks @jimw567 and @kristopk for suggestion. closing this as no further upadates from filer for this issue. 

Hi @ramonn76 , please post query into AWS forum if you still face this issue.

-Heera

closed 2018-06-29 by heeran-xilinx