Xilinx/SDAccel_Examples #36: Using Multiple DDRs

aeeshms, 2018-09-04:
Hi
Setting up the multiple DDR using the command below in xocc compiler and linker setting
-max_memory_ports all --sp cnn_1.m_axi_gmem0:bank0 --sp cnn_1.m_axi_gmem1:bank1 
gives a error:

"make incremental 
/opt/Xilinx/SDx/2017.1/bin/xocc -t hw_emu --platform xilinx:kcu1500:4ddr-xpr:4.0 --save-temps  --report estimate -c -k cnn -g --messageDb cnn_GOOD/cnn.mdb -I"../src" --xp misc:solution_name=_xocc_compile_cnn_GOOD_cnn -max_memory_ports all --sp cnn_1.m_axi_gmem0:bank0 --sp cnn_1.m_axi_gmem1:bank1 -o"cnn_GOOD/cnn.xo" "../src/cnn_convolution.cpp"

/opt/Xilinx/SDx/2017.1/bin/unwrapped/lnx64.o/xocc: invalid option -- 'm'
"
The sample design fails and gives a error

ted-xie, 2018-09-18:
You may have just formatted your command incorrectly. In past experience, the `--xp` options need to be quote escaped, i.e. `--xp "misc:..."` instead of `--xp misc:...`.

aeeshms, 2018-09-18:
correct command:
--max_memory_ports cnn --sp cnn_1.m_axi_gmem0:bank0 --sp cnn_1.m_axi_gmem1:bank1  --sp cnn_2.m_axi_gmem0:bank0 --sp cnn_2.m_axi_gmem1:bank1   --sp cnn_3.m_axi_gmem0:bank0 --sp cnn_3.m_axi_gmem1:bank1    --sp cnn_4.m_axi_gmem0:bank0 --sp cnn_4.m_axi_gmem1:bank1

heeran-xilinx, 2018-11-21:
Closing this as correct commands are there in latest Git master.

closed 2018-11-21 by heeran-xilinx