Xilinx/SDAccel_Examples #38: Runtime include errors

definelicht, 2018-10-08:
This commit seems to have broken all examples: 3b7e906d36d30ebb7b0e24fccdc04427acb87b31

Compiling any example gives the following error because XILINX_XRT is not set (also there is no possible value of XILINX_XRT that is correct, since it seems to assume a different directory structure from what's in SDAccel 2018.2):

```
% XILINX_SDX=/opt/Xilinx/SDx/2018.2 make exe
mkdir -p ./xclbin
/opt/Xilinx/SDx/2018.2/bin/xcpp -I /include/ -I//opt/Xilinx/SDx/2018.2/Vivado_HLS/include/ -O0 -g -Wall -fmessage-length=0 -std=c++14 -I../../..//libs/xcl2 src/host.cpp ../../..//libs/xcl2/xcl2.cpp -o 'vadd' -lOpenCL -lpthread -lrt -lstdc++ -L/lib/ 
In file included from src/host.cpp:29:0:
../../..//libs/xcl2/xcl2.hpp:47:10: fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../../..//libs/xcl2/xcl2.cpp:33:0:
../../..//libs/xcl2/xcl2.hpp:47:10: fatal error: CL/cl2.hpp: No such file or directory
 #include <CL/cl2.hpp>
          ^~~~~~~~~~~~
compilation terminated.
```

What gives?

aharishkumar, 2018-10-08:
Hi @definelicht , please source the following,

source /proj/xbuilds/2018.2_daily_latest/xbb/xrt/packages/<ubuntu/centos>/opt/xilinx/xrt/setup.<sh/csh>

This sets the XILINX_XRT env variable and this issue must not be seen.

Thanks,
 Harish.

definelicht, 2018-10-10:
Hi @aharishkumar, thanks for the response.
I did not find this information in any of the README files. I would suggest that you update the documentation to make this more clear :-)

vird, 2019-05-11:
apt install opencl-headers
That helps me