Xilinx/SDAccel_Examples #39: XOCC crash

ted-xie, 2018-11-08:
I've followed the instructions to install xocc on a t2.nano instance and came across this error:

```
$ xocc
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
/opt/Xilinx/SDx/2018.2.op2258646/bin/loader: line 194:  1869 Aborted                 "$RDI_PROG" "$@"
```

I am following the instructions [here](https://github.com/Xilinx/SDAccel_Examples/wiki/Create,-configure-and-test-an-AWS-F1-instance#configuring-the-sdaccel-environment). These instructions used to work for me, but not anymore. Thoughts?

ted-xie, 2018-11-09:
For setup, I am running this script ([link](https://github.com/ted-xie/aws_fpga_init/blob/master/init.sh)). The commands are directly copied from the "Getting started page" for AWS F1.

heeran-xilinx, 2018-11-12:
Hi @ted-xie ,

Could you please post your AWS/F1 setup related queries to F1 Forum?
https://forums.aws.amazon.com/forum.jspa?forumID=243&start=0

I hope you may get faster response there.

-Heera

heeran-xilinx, 2018-11-21:
Closing this as requested file F1 related issues to F1 Forum

closed 2018-11-21 by heeran-xilinx