Xilinx/SDAccel_Examples #43: How to give user defined input sequences for alignment?

manukondaraju, 2019-04-02:
The example doesn't show what are the input sequences and what is the output sequences. Looks like the code generates two sequences. But I want to give my own input sequences for alignment. How to do that?

heeran-xilinx, 2019-04-02:
Hi @manukondaraju ,
Could you please elaborate more? For which example you are talking about? 

-Heera

manukondaraju, 2019-04-02:
Sure @heeran-xilinx ..I am running the smithwaterman algorithm in acceleration directory. I want to know how to give two input sequences for alignment in this code.

manukondaraju, 2019-04-04:
Hi @heeran-xilinx @heeranand  Can I get the solution for the problem?

manukondaraju, 2019-04-04:
Hi @heeran-xilinx  can you please help me through this issue?

kaliudayxilinx, 2019-04-08:
Hi @manukondaraju 

This code generates ref.txt (default) sequence using genSeq() method. Files are not dumped default.
You can use -f or --sample-file option to provide file of your choice. 

There is a bug in the code which restricts you to pass a file,

Line 77:main.cpp --> Change it to (argc < 2)

Also I would recommend you to print the default "ref.txt" which they are creating, pass on the input based on the format followed in auto generated ref.text.

Example Usage:
./smithwaterman xclbin/krnl_smithwaterman.sw_emu.xilinx_u200_xdma_201830_1.xclbin -f input.txt -vm 1

Thanks
Kali

manukondaraju, 2019-04-09:
Hi @kaliudayxilinx 

After running the code in SDAccel toolchain I don't see any ref.txt file being generated and so I am unable to print the contents of the file to see the input format. If possible can you do the same for me and   let me know the input format here, so that I can give my own inputs to the program.

kaliudayxilinx, 2019-04-09:
Hi @manukondaraju 

As I mentioned previously, input/output files are not dumped by default.

In src/smithwaterman.cpp file Look at the line numbers 461 to 472. If input file is specified by user it will be picked by in verify mode (-vm 1 is must). If no input file is provided then it goes into else condition where it generates the read-ref samples. 

In both the cases input data is stored in "input" buffer.

Kindly print this buffer to know the format they follow.  Kindly do some exploration of the example provided in order to understand the format provided. It's crowd sourced example support is limited.

``
for (int i = 0; i < hwBlockSize * PACKEDSZ;i++) {
         printf("Data: %c / %d \n", input[i], input[i]);
}
``

Thanks
Kali

manukondaraju, 2019-04-09:
Hi @kaliudayxilinx 

I have tried printing the input buffer to see the format, but it prints garbage values like this.

Data: � / -2077771049 
Data: � / -1144524558 
Data: � / 1197032080 
Data: ] / -1753037987 
Data: � / -1735077968 
Data: p / -2024507024 e format they follow. Kindly do some exploration of the example provided in order to understand the format provided. It's crowd sourced example support is limited.
Data: � / -1249063973 
Data: � / 1375341750 

Thanks
Raju

manukondaraju, 2019-04-09:
Hi @kaliudayxilinx 

I have tried printing the input buffer to see the format, but it prints garbage values like this.

Data: � / -2077771049
Data: � / -1144524558
Data: � / 1197032080
Data: ] / -1753037987
Data: � / -1735077968
Data: p / -2024507024 
Data: � / -1249063973
Data: � / 1375341750

manukondaraju, 2019-04-11:
Hi @kaliudayxilinx , @heeran-xilinx  

I tried walking through the code about how they are generating the sequences but could not understand much on that. But seeing the input format says that is not the correct input as it is showing garbage values. Can you please help on this?

heeran-xilinx, 2020-07-16:
Closing this issue as this repository is only for older released version of tool and no more support is available.
Please use latest Repository: https://github.com/Xilinx/Vitis_Accel_Examples for Xilinx latest Tool chain.

closed 2020-07-16 by heeran-xilinx