Xilinx/SDAccel_Examples #47: how to generate waveform in example "getting_started/kernel_gmem/burst_rw_c" ?

xiaoliang201188, 2019-05-07:
There was no "*.wdb" file when I used "make all TARGET=hw_emu DEVICE=xilinx_u200_xdma_201820_1" to build .

xiaoliang201188, 2019-05-08:
[image]

kapil2099, 2019-05-14:
Hi , sorry for the late reply.

To only save .wdb file 
Please include following in your sdaccel.ini :

[Emulation]
launch_waveform=batch 

To save .wdb as well as view it in GUI
Please include following in your sdaccel.ini :

[Emulation]
launch_waveform=gui

heeran-xilinx, 2019-05-15:
HI @xiaoliang201188 ,
Please refer this example which dump .wdb file. I has sdaccel.ini containing required information for the same.

-Heera

closed 2019-05-15 by heeran-xilinx

xiaoliang201188, 2019-05-15:
Solved , Thank you!