Xilinx/SDAccel_Examples #49: hw_emu profiled data is different with hw in example "too_many_cu_c"

xiaoliang201188, 2019-05-24:
"SDAccel_Examples-master\SDAccel_Examples-master\getting_started\clk_freq\too_many_cu_c"
[image]
hw: Compute Unit vadd_BAD_1 execution counts are 8，vadd_BAD_2 to vadd_BAD_8 are not be stalled。
[image]
hw_emu: Worked as expected of host
[image]

heeran-xilinx, 2019-05-24:
Hi @xiaoliang201188 ,
Thanks for reporting this. 
Actually Kernel execution time is so small "0.047 ms", that by the time when host schedule the next kernel call, previous kernel call is already over. As a result, Runtime is allocating the same compute unit again as it is ready (not busy).

Whereas in HW emulation (as it runs on RTL simulation), kernel execution model runs very slow and host can easily schedule the operations to all Compute units as each compute unit takes significant time to finish (as RTL simulation takes much longer time to finish). 

I hope this clarify  your doubts.
-Heera

xiaoliang201188, 2019-05-24:
> Hi @xiaoliang201188 ,
> Thanks for reporting this.
> Actually Kernel execution time is so small "0.047 ms", that by the time when host schedule the next kernel call, previous kernel call is already [...]

> Hi @xiaoliang201188 ,
> Thanks for reporting this.
> Actually Kernel execution time is so small "0.047 ms", that by the time when host schedule the next kernel call, previous kernel call is already [...]

Thanks for your reply. If I want call all kernels execute parallelly, how to modify the host code in this example ？

heeran-xilinx, 2019-05-24:
Hi @xiaoliang201188 ,
Triggering kernel from host to device is always sequential, Triggering kernel from host to device needs some setup time. If this setup time is similar in the range of kernel execution time, you would not be able to run them concurrently. It is because, when host will try to configure the 2nd kernel, first is already finished and will be re-used for 2nd kernel call as well. So you will always noticed that same compute unit is always getting used. So to see the effect of running compute unit concurrently, you need to make sure kernel execution time should be much higher than setup time.
You can refer K-means (under acceleration) examples, where multiple computes are running concurrently in real board as well.

-Heera

xiaoliang201188, 2019-05-24:
OK，thanks

closed 2019-05-24 by xiaoliang201188