Xilinx/SDAccel_Examples #50: Issues when installing SHA on aws FPGA

xcwanAndy, 2019-06-05:
Hi, I'm new about aws fpga and want to try sha on the board.

I'm following the procedures shown here: https://github.com/Xilinx/SDAccel_Examples/tree/2018.2/security/sha1

I've used **_FPGA Developer AMI v1.5_** to launch a **f1-2xlarge** instance on aws. However, as I'm trying to first **Compiling for Application Emulation**, I met the following messages:

```
[centos@ip-172-31-9-50 sha1]$ make TARGETS=sw_emu all
mkdir -p ./xclbin
/opt/Xilinx/SDx/2018.2.op2258646/bin/xocc -t sw_emu --platform xilinx_vcu1525_dynamic --save-temps  --xp "param:compiler.preserveHlsOutput=1" --xp "param:compiler.generateExtraRunData=true" -c -k dev_sha1_update -I'src' -o'xclbin/krnl_sha1.sw_emu.xilinx_vcu1525_dynamic.xo' 'src/krnl_clSha1.cl'

****** xocc v2018.2_AR71715_op (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

Attempting to get a license: ap_opencl
Feature available: ap_opencl
INFO: [XOCC 60-585] Compiling for software emulation target
Running SDx Rule Check Server on port:40026
ERROR: [XOCC 60-705] No platform was found that matches 'xilinx_vcu1525_dynamic'. Make sure that the platform is specified correctly and that valid license is installed. The valid platforms supported by the license found are:

ERROR: [XOCC 60-587] Failed to add a platform: specified platform xilinx_vcu1525_dynamic is not found
ERROR: [XOCC 60-600] Kernel compile setup failed to complete
ERROR: [XOCC 60-592] Failed to finish compilation
make: *** [xclbin/krnl_sha1.sw_emu.xilinx_vcu1525_dynamic.xo] Error 1
```

Can anyone help me fix this please?

heeran-xilinx, 2019-06-07:
Hi @xcwanAndy ,
Please specify the absolute path of xpfm file to DEVICE.

make TARGETS=sw_emu all DEVICE=<xpfm full path>

-Heera

xcwanAndy, 2019-06-09:
@heeran-xilinx Thanks a lot for your reply! Now I face another problem.

```
% sudo ./sha1 -t 20 -p Xilinx -k ./xclbin/krnl_sha1.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin
DEVICES=/home/centos/xinchen/aws-fpga/SDAccel/aws_platform/xilinx_aws-vu9p-f1-04261818_dynamic_5_0/xilinx_aws-vu9p-f1-04261818_dyn
amic_5_0.xpfm DEVICES=$AWS_PLATFORM
INFO: [main.cpp:343] TIME: [Sun Jun  9 14:03:17 2019] Xilinx SHA1 Example Application
INFO: [cmdlineparser.cpp:227] TIME: [Sun Jun  9 14:03:17 2019] Using default key: --kernel-file
INFO: [main.cpp:374] TIME: [Sun Jun  9 14:03:17 2019] Platform: Xilinx, Device:
INFO: [main.cpp:375] TIME: [Sun Jun  9 14:03:17 2019] Kernel FP: DEVICES=/home/centos/xinchen/aws-fpga/SDAccel/aws_platform/xilinx_aws-vu9p-f1-04261818_dynamic_5_0/xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xpfm
INFO: [main.cpp:376] TIME: [Sun Jun  9 14:03:17 2019] Runners: 2 (512.031250 MB)
INFO: [main.cpp:377] TIME: [Sun Jun  9 14:03:17 2019] ZMQ PORT: 5010
INFO: [main.cpp:378] TIME: [Sun Jun  9 14:03:17 2019] Running for [20.000000] seconds...
ERROR: Library libxdp.so not found! XILINX_XRT not set
ERROR: XILINX_XRT must be set
ERROR: no platforms available or OpenCL install broken
Aborted
```

And the XILINX_XRT is:
```
% echo $XILINX_XRT
/opt/xilinx/xrt
```

The address of "libxdp.so" is:
```
% sudo find / -name "libxdp.so*"
/opt/xilinx/xrt/lib/libxdp.so
```

I think the problem here is because I have changed Xilinx Runtime version due to the messages I saw in the past installation procedure.

 I followed the instruction here "https://github.com/aws/aws-fpga/blob/master/SDAccel/docs/XRT_installation_instructions.md#installing-xilinx-runtime-xrt-20182_xdfrc5" and installed xrt_201802.2.1.0_7.5.1804-xrt.rpm and xrt_201802.2.1.0_7.5.1804-aws.rpm. 

And then there exists /opt/Xilinx/SDx/2018.2.op2258646 directory in the host. But I didn't see any "setup.sh" file in it to configure the new xilinx runtime.

Would you please give me some suggestions? Thanks a lot!

xcwanAndy, 2019-06-10:
@heeran-xilinx  Hi Heeran! Now occurs another problem.

```
sh-4.2# ./sha1 -t 20 -p Xilinx -k ./xclbin/krnl_sha1.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin DEVICES=/home/centos/xinchen/aws-fpga/SD
Accel/aws_platform/xilinx_aws-vu9p-f1-04261818_dynamic_5_0/xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xpfm
INFO: [main.cpp:343] TIME: [Mon Jun 10 16:03:32 2019] Xilinx SHA1 Example Application
INFO: [main.cpp:374] TIME: [Mon Jun 10 16:03:32 2019] Platform: Xilinx, Device:
INFO: [main.cpp:375] TIME: [Mon Jun 10 16:03:32 2019] Kernel FP: ./xclbin/krnl_sha1.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin
INFO: [main.cpp:376] TIME: [Mon Jun 10 16:03:32 2019] Runners: 2 (512.031250 MB)
INFO: [main.cpp:377] TIME: [Mon Jun 10 16:03:32 2019] ZMQ PORT: 5010
INFO: [main.cpp:378] TIME: [Mon Jun 10 16:03:32 2019] Running for [20.000000] seconds...
xclProbe found 1 FPGA slots with xocl driver running
Dev:
Match: xilinx_aws-vu9p-f1-04261818_dynamic_5_0
Segmentation fault
```

Does it mean that the SHA program can not be configured on aws f1-2xlarge? Because "https://github.com/Xilinx/SDAccel_Examples/tree/2018.2/security/sha1#3-software-and-system-requirements" mentions the requirement of hardware board, and aws f1-2xlarge doesn't provide these two types of boards.

But from the hardware configuration, I think the FPGA on f1-2xlarge has better resources than the required two types.

Please help me with this issue. Thanks a lot for your kindness!

xcwanAndy, 2019-06-10:
@heeran-xilinx Hello Heeran! Thank you for your kindly reply. Now I face a strange thing while running the SHA program, which is shown below:

```
[centos@ip-172-31-92-105 sha1]$ make check TARGETS=hw_emu DEVICES=/home/centos/xinchen/aws-fpga/SDAccel/aws_platform/xilinx_aws-vu9p-f1-04261818_dynamic_5_0/xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xpfm all
cp -rf ./xclbin/xilinx_aws-vu9p-f1-04261818_dynamic_5_0/emconfig.json .
XCL_EMULATION_MODE=hw_emu ./sha1 -t 20 -p Xilinx -k ./xclbin/krnl_sha1.hw_emu.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin
INFO: [main.cpp:343] TIME: [Mon Jun 10 16:38:24 2019] Xilinx SHA1 Example Application
INFO: [main.cpp:374] TIME: [Mon Jun 10 16:38:24 2019] Platform: Xilinx, Device:
INFO: [main.cpp:375] TIME: [Mon Jun 10 16:38:24 2019] Kernel FP: ./xclbin/krnl_sha1.hw_emu.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin
INFO: [main.cpp:376] TIME: [Mon Jun 10 16:38:24 2019] Runners: 2 (512.031250 MB)
INFO: [main.cpp:377] TIME: [Mon Jun 10 16:38:24 2019] ZMQ PORT: 5010
INFO: [main.cpp:378] TIME: [Mon Jun 10 16:38:24 2019] Running for [20.000000] seconds...
ERROR: Unable to open profile report for writing
xclProbe found 1 FPGA slots with xocl driver running
Dev:
Match: xilinx_aws-vu9p-f1-04261818_dynamic_5_0
INFO: [SDx-EM 01] Hardware emulation runs simulation underneath. Using a large data set will result in long simulation times. It is recommended that a small dataset is used for faster execution. This flow does not use cycle accurate models and hence the performance data generated is approximate.
INFO: FPGA Start
INFO: Starting Timer
INFO: Test complete ran for 20 seconds! Stopping...
INFO: Job Size: 1 MB
INFO: Jobs Processed: 0
INFO: Rate = 0 MB/s
```

The software emulation works quite well and I got a reasonable rate number. However, it seems that the hardware emulation is incorrect.

And after I loaded the image that is generated onto the FPGA board, the running result is:
```
sh-4.2# ./sha1 -t 20 -p Xilinx -k ./xclbin/krnl_sha1.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin DEVICES=/home/centos/xinchen/aws-fpga/SD
Accel/aws_platform/xilinx_aws-vu9p-f1-04261818_dynamic_5_0/xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xpfm
INFO: main.cpp:343 TIME: https://forums.aws.amazon.com/ Xilinx SHA1 Example Application
INFO: main.cpp:374 TIME: https://forums.aws.amazon.com/ Platform: Xilinx, Device:
INFO: main.cpp:375 TIME: https://forums.aws.amazon.com/ Kernel FP: ./xclbin/krnl_sha1.hw.xilinx_aws-vu9p-f1-04261818_dynamic_5_0.xclbin
INFO: main.cpp:376 TIME: https://forums.aws.amazon.com/ Runners: 2 (512.031250 MB)
INFO: main.cpp:377 TIME: https://forums.aws.amazon.com/ ZMQ PORT: 5010
INFO: main.cpp:378 TIME: https://forums.aws.amazon.com/ Running for http://20.000000 seconds...
xclProbe found 1 FPGA slots with xocl driver running
Dev:
Match: xilinx_aws-vu9p-f1-04261818_dynamic_5_0
Segmentation fault
```

Does this means that I have set the hardware environment wrongly?

Would you please give me some suggestions to settle the issue down? Thanks.

heeran-xilinx, 2020-07-16:
Closing this issue as Git repository is only for older released version of tool and no more support is available.
Please use latest Repository: https://github.com/Xilinx/Vitis_Accel_Examples for Xilinx latest Tool chain.

closed 2020-07-16 by heeran-xilinx