Xilinx/SDAccel_Examples #54: SDAccel_Examples/getting_started/rtl_kernel/rtl_streaming_free_running

saitej25, 2019-08-05:
Example Design giving errors when I tried to build
make all TARGET=hw_emu DEVICE=xilinx_u200_qdma_201910_1 check

logs:
/opt/xilinx/SDx/2019.1/bin/xcpp -I../../..//libs/xcl2 -pthread -I/scratch/aberroc/xrt_2.1_patched/opt/xilinx/xrt/include/ -I/opt/xilinx/Vivado/2019.1/include/ -Wall -O0 -g -std=c++14 -fmessage-length=0 ../../..//libs/xcl2/xcl2.cpp src/host.cpp  -o 'vadd_stream'  -L/scratch/aberroc/xrt_2.1_patched/opt/xilinx/xrt/lib/ -lOpenCL -lpthread -lrt -lstdc++ In file included from /usr/include/c++/5/thread:39:0,                  from src/host.cpp:33: /usr/include/c++/5/functional: In instantiation of ‘struct std::_Bind_check_arity<int (*)(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*), _cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*>’: /usr/include/c++/5/functional:1538:12:   required from ‘struct std::_Bind_simple_helper<int (*&)(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*), _cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*>’ /usr/include/c++/5/functional:1552:5:   required by substitution of ‘template<class _Callable, class ... _Args> typename std::_Bind_simple_helper<_Func, _BoundArgs>::__type std::__bind_simple(_Callable&&, _Args&& ...) [with _Callable = int (*&)(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*); _Args = {_cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*}]’ /usr/include/c++/5/thread:137:59:   required from ‘std::thread::thread(_Callable&&, _Args&& ...) [with _Callable = int (*&)(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*); _Args = {_cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*}]’ src/host.cpp:181:26:   required from here /usr/include/c++/5/functional:1410:7: error: static assertion failed: Wrong number of arguments for function        static_assert(sizeof...(_BoundArgs) == sizeof...(_Args),        ^ /usr/include/c++/5/functional: In instantiation of ‘struct std::_Bind_simple<int (*(_cl_stream*, int*, unsigned int, stream_xfer_req*, int*))(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*)>’: /usr/include/c++/5/thread:137:59:   required from ‘std::thread::thread(_Callable&&, _Args&& ...) [with _Callable = int (*&)(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*); _Args = {_cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*}]’ src/host.cpp:181:26:   required from here /usr/include/c++/5/functional:1505:61: error: no type named ‘type’ in ‘class std::result_of<int (*(_cl_stream*, int*, unsigned int, stream_xfer_req*, int*))(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*)>’        typedef typename result_of<_Callable(_Args...)>::type result_type;                                                              ^ /usr/include/c++/5/functional:1526:9: error: no type named ‘type’ in ‘class std::result_of<int (*(_cl_stream*, int*, unsigned int, stream_xfer_req*, int*))(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*)>’          _M_invoke(_Index_tuple<_Indices...>)          ^ /usr/include/c++/5/functional: In instantiation of ‘struct std::_Bind_check_arity<int (*)(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*), _cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*>’: /usr/include/c++/5/functional:1538:12:   required from ‘struct std::_Bind_simple_helper<int (*&)(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*), _cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*>’ /usr/include/c++/5/functional:1552:5:   required by substitution of ‘template<class _Callable, class ... _Args> typename std::_Bind_simple_helper<_Func, _BoundArgs>::__type std::__bind_simple(_Callable&&, _Args&& ...) [with _Callable = int (*&)(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*); _Args = {_cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*}]’ /usr/include/c++/5/thread:137:59:   required from ‘std::thread::thread(_Callable&&, _Args&& ...) [with _Callable = int (*&)(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*); _Args = {_cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*}]’ src/host.cpp:193:26:   required from here /usr/include/c++/5/functional:1410:7: error: static assertion failed: Wrong number of arguments for function        static_assert(sizeof...(_BoundArgs) == sizeof...(_Args),        ^ /usr/include/c++/5/functional: In instantiation of ‘struct std::_Bind_simple<int (*(_cl_stream*, int*, unsigned int, stream_xfer_req*, int*))(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*)>’: /usr/include/c++/5/thread:137:59:   required from ‘std::thread::thread(_Callable&&, _Args&& ...) [with _Callable = int (*&)(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*); _Args = {_cl_stream*&, int*, unsigned int&, stream_xfer_req*, int*}]’ src/host.cpp:193:26:   required from here /usr/include/c++/5/functional:1505:61: error: no type named ‘type’ in ‘class std::result_of<int (*(_cl_stream*, int*, unsigned int, stream_xfer_req*, int*))(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*)>’        typedef typename result_of<_Callable(_Args...)>::type result_type;                                                              ^ /usr/include/c++/5/functional:1526:9: error: no type named ‘type’ in ‘class std::result_of<int (*(_cl_stream*, int*, unsigned int, stream_xfer_req*, int*))(_cl_device_id*, _cl_stream*, void*, long unsigned int, long unsigned int, stream_xfer_req*, int*)>’          _M_invoke(_Index_tuple<_Indices...>)          ^ Makefile:95: recipe for target 'vadd_stream' failed make: *** [vadd_stream] Error 1SDAccel_Examples/getting_started/rtl_kernel/rtl_streaming_free_running

rajatsarkari, 2019-08-06:
Hi @saitej25 ,
Could you please give the complete path to the .xpfm file in the DEVICE section?
For example: make all TARGET=hw_emu DEVICE=<PATH_TO_DEVICE>/xilinx_u200_qdma_201910_1/xilinx_u200_qdma_201910_1.xpfm check
Regards,
Rajat Sarkari

jstonge640, 2019-09-23:
I am having the same problem when trying to compile `rtl_streaming_free_running` on the nimbix cloud.  I tried using the absolute path as suggested and I ran into another error (see snippet below).  Please see [vivado.log](https://github.com/Xilinx/SDAccel_Examples/files/3642115/vivado.log) for the complete log file.

`Creating apsys_0.xml
ERROR: [CFGEN 83-2284] No stream resources found that can accomodate compute unit "myadder1_1"
Error generating design file for apsys_0.xml
command: /opt/xilinx/SDx/2019.1/bin/cfgen    -r /data/SDAccel_Examples/getting_started/rtl_kernel/rtl_streaming_free_running/_x/link/sys_link/_sds/.cdb/xd_ip_db.xml -o apsys_0.xml
Error creating intermediate design file, exiting
INFO: [XOCC 60-1442] [12:29:57] Run xocc_link: Step sdx_link: Failed
Time (s): cpu = 00:00:07 ; elapsed = 00:00:09 . Memory (MB): peak = 563.199 ; gain = 0.000 ; free physical = 63548 ; free virtual = 196247
ERROR: [XOCC 60-661] xocc link run 'xocc_link' failed
ERROR: [XOCC 60-626] Kernel link failed to complete
ERROR: [XOCC 60-703] Failed to finish linking
Makefile:90: recipe for target 'xclbin/myadder1.hw_emu.xilinx_u200_xdma_201830_2.xclbin' failed
make: *** [xclbin/myadder1.hw_emu.xilinx_u200_xdma_201830_2.xclbin] Error 1`

heeran-xilinx, 2019-09-23:
rtl_streaming_free_running example is supported only by QDMA platform. 
But as per log it looks like XDMA platform is used:
Please see the supported platform list in readme:
https://github.com/Xilinx/SDAccel_Examples/blob/master/getting_started/rtl_kernel/rtl_streaming_free_running/README.md

jstonge640, 2019-09-23:
> rtl_streaming_free_running example is supported only by QDMA platform.
> But as per log it looks like XDMA platform is used:
> Please see the supported platform list in readme:
> https://github.com/Xilinx/SDAccel_Examples/blob/master/getting_started/rtl_kernel/rtl_streaming_free_running/README.md

My mistake.  Thanks!  It appears Nimbix doesn't have any QDMA .xpfm platforms installed so I will reach out to them, unless there's another location for that file?

jstonge640, 2019-09-23:
I believe that the absolute path is still required, so shouldn't this issue remain open?  The `pinfo.json does not exist!` error still persists when trying to use just the device name.

saitej25, 2019-09-27:
> I believe that the absolute path is still required, so shouldn't this issue remain open? The `pinfo.json does not exist!` error still persists when trying to use only the device name.

git issues in this repo are a joke anyway.

heeran-xilinx, 2019-09-29:
Re-opening this issue to remove the restriction of absolute path. Which will be addressed in next release. For now, please use absolute path as work-around.

shivrajatrialogic, 2019-10-02:
I am also facing this issue.
Also the Host.cpp source code doesnt compile.
Throws multiple error:
1. error: static assertion failed: Wrong number of arguments for pointer-to-member
2. functional:1395:9: error: no type named \u2018type\u2019 in \u2018class std::result_of<int (*(_cl_stream*, int*, unsigned int, stream_xfer_req*, int*))(_cl_device_id*, _cl_stream*, const void*, long unsigned int, long unsigned int, stream_xfer_req*, int*)>\u2019 _M_invoke(_Index_tuple<_Indices...>)

I get these errors when  i make :Follows
make check TARGETS=hw_emu DEVICES=<Board>

heeran-xilinx, 2019-11-08:
This issue has been addressed in new Git repository:
https://github.com/Xilinx/Vitis_Accel_Examples

This repo support Xilinx new Tool Vitis 2019.2

closed 2019-11-08 by heeran-xilinx