Xilinx/SDAccel_Examples #56: Host streaming example for OpenCL

scarlehoff, 2019-10-19:
I can't help but noticing that all examples for host streaming are for Cpp kernels. Is it not possible to achieve the same functionality with OpenCL kernels?

I'm thinking for instance of `host/streaming_host_bandwidth` or even `host/streaming_simple`.

heeran-xilinx, 2019-10-19:
Hi @scarlehoff , 
Streaming kernel support is  provided for HLS & RTL kernel only. Please post your query to SDAccel forum to check if OpenCL kernels are planned to support in future or not.

heeran-xilinx, 2020-07-16:
Closing this as this repository is only for older released version of tool and no more support is available.
Please use latest Repository: https://github.com/Xilinx/Vitis_Accel_Examples for Xilinx latest Tool chain.

closed 2020-07-16 by heeran-xilinx