Xilinx/SDAccel_Examples #59: EDGE_COMMON_SW variable is not set

jrjbertram, 2020-07-16:
Hi there,

Seeing this error when attempting to build the hello world example for zcu102 board for software emulation.  (Same thing for aarch32.)  Not sure what I'm missing.

- J.

```
(base) adclab@solo:~/Vitis_Accel_Examples/hello_world$ make all TARGET=sw_emu DEVICE=xilinx_zcu102_base_202010_1 HOST_ARCH=aarch64
utils.mk:61: *** EDGE_COMMON_SW variable is not set, please set correctly and rerun.  Stop.
```

heeran-xilinx, 2020-07-16:
Hi @jrjbertram ,
For Vitis project a completely new Repository is created  https://github.com/Xilinx/Vitis_Accel_Examples
Please use the same.

SDAccel Examples repository is there for older SDx product support only.

closed 2020-07-16 by heeran-xilinx

jrjbertram, 2020-07-16:
Hi, consider adding a brief statement to the README.md to this effect.

heeran-xilinx, 2020-07-16:
Thanks @jrjbertram  for suggestion. I added a Note at the start of the README for the same.
Note: Please do not use this repository for Xilinx Latest Vitis Tool Chain. A entire new Repository is moved here: https://github.com/Xilinx/Vitis_Accel_Examples for Xilinx new Vitis Tool chain. Use SDAccel Examples repository for Xilinx previously released product SDAccel or SDx.